wagiminator/Development-Boards #2: CH559 No device detected on PC

cuishuang413, 2024-02-13:
Hey all,

I just built a CH559T board and verified that all soldering are solid and correct. No device is detected at all when plug into a USB port. I'm using Windows and got the CH375 driver installed. I got the CH559T chip from Aliexpress. Anyone got any ideas?

Thanks!

wagiminator, 2024-02-13:
Hi, please try the following:
Connect pin P46 to ground (GND) with a jumper wire, then connect the board via USB to your PC without pressing the BOOT button.

cuishuang413, 2024-02-13:
> Hi, please try the following: Connect pin P46 to ground (GND) with a jumper wire, then connect the board via USB to your PC without pressing the BOOT button.

Hi there!

Thanks for your reply. I just tried it but no dice. I put a scope on the D+ and D- pins, there is no activity at all. Does all CH559T come with bootloaders? It feels like the chips I got doesn't have a bootloader in it.

Thanks!

wagiminator, 2024-02-13:
As far as I know, they all have a bootloader integrated, but unfortunately the documentation and data sheets are so poor that I can't say for sure. If you can't detect any activity, it doesn't necessarily mean that the chip doesn't have a bootloader. Maybe it just isn't activated. My CH559T are from LCSC, but I have many other chips from Aliexpress that also worked without any problems. So unfortunately, I can't tell you what the problem is at the moment. Do you have the link to Aliexpress? Maybe I'll order a few and do some tests.

cuishuang413, 2024-02-13:
Here is the link where I ordered mine: 
https://a.aliexpress.com/_mOYVnzc
Thank you so much for helping! I'm really interested in exploring these chips as a potential solution for a few open source project of mine.

Not sure if this is any helpful, I measured the levels on all the pins. The 3.3V is generated from the internal LDO fine. All pins except D+, D-, RXD, TXD are high at about 3.3V. The said four pins low at 0V.

Thanks!

wagiminator, 2024-02-14:
Unfortunately the item is not available in my country.

cuishuang413, 2024-02-14:
I will order from LCSC and try again. I got both CH559L and CH559T. Both didn't work. I suspect it's the chip problem. Thanks for the help!