wagiminator/Development-Boards #6: Using latest SDCC doesn't fit in CH554E_HostDevBoard

akuker, 2024-07-06:
### Issue
The current CH554E_HostDevBoard doesn't fit in memory when its compiled as-is. 

When the example is built on Ubuntu 22 with sdcc 4.2.0 using the `make` command, the following output is observed:

```
Compiling host_analyzer.c ...
Compiling include/debug.c ...
Compiling include/delay.c ...
Compiling include/usb_host.c ...
Building host_analyzer.ihx ...
?ASlink-Error-Could not get 46 consecutive bytes in internal RAM for area DSEG.

?ASlink-Error-Insufficient ROM/EPROM/FLASH memory.
make: *** [makefile:50: host_analyzer.ihx] Error 1
```

This is similar to the issue reported on the ch554_sdcc project: https://github.com/Blinkinlabs/ch554_sdcc/issues/48

###Workaround 
As a temporary work-around, the `define printf printf_tiny` line can be enabled in `include/debug.h`. However, this stripped down version of printf appears to just dump the string without formatting.

Another thing that reduced the amount of flash used was to add the `--nogcse` option to the `CFLAGS` list in the makefile. However, this change alone did not shrink the code enough to fit in the ch554e.
Reference: https://sdcc-user.narkive.com/2CPmObSg/emergency-help-linker-memory-error-with-dseg

wagiminator, 2024-07-06:
Oh yes, you are right, thanks for the hint! It doesn't work on my current computer (SDCC v4.0.0) either. I will have to investigate a bit when I find the time. Until then, the precompiled binary should at least work.

akuker, 2024-07-08:
Splitting up usb_host seemed to help: https://github.com/wagiminator/Development-Boards/compare/main...akuker:Development-Boards:main

I also made some updates to the makefile to put the build files in an `obj` directory and added auto-include functionality. 

@wagiminator - would you like a PR with these changes?

wagiminator, 2024-07-08:
Strange! If it works, then you can make a PR.

akuker, 2024-07-09:
Trying to understand why it works. I spot checked the generated assembly, and it looks the same. But in the .map file, different amounts of CSEC memory are being allocated when all functions are in the same file. I don't have my board with me, but I'll do a bunch more testing this weekend before I submit a PR.

[image]