wagiminator/Development-Boards #9: CH32X035 F7P6 Dev Board fried

R8bert, 2024-10-19:
If I increase the default Power Delivery (PD) voltage from the configuration file beyond the specified 9000 mV, my microcontroller becomes damaged. The HT7533 is overheating and that is it, my mcu is fryed, after that my pc detects that the usb port is shorted if i plug the dev board in.  I have tried this with 2 boards having the same result 
[image]

I am doing something wrong?

wagiminator, 2024-10-19:
Hi, strange. Did you make sure that the input capacitor on the LDO is rated for 35V? Did you have a larger load connected to the output of the board? Is your LDO suitable for at least 20V input voltage (there are different HT7533 out there)?

R8bert, 2024-10-19:
> Hi, strange. Did you make sure that the input capacitor on the LDO is rated for 35V? Did you have a larger load connected to the output of the board? Is your LDO suitable for at least 20V input voltage (there are different HT7533 out there)?

Hi Stefan,
Thank you for your quick reply!

I think the LDO capacitors are not rated for 35V input. I used them from another project (https://www.aliexpress.com/item/1005005690927737.html). I don't think these capacitors can handle this voltage, but I thought that when a capacitor exceeds its rated voltage, it should short to ground, preventing damage to the microcontroller. Shouldn't the microcontroller be protected from overvoltage ? However, that apparently didn't happen, and the MCU was still fried.

Can you help clarify this for me? I am still a beginner and still learning this stuff:))

wagiminator, 2024-10-20:
If the capacitor is overloaded, it typically becomes conductive and causes a short circuit on the board. However, you won’t know if the MCU is also damaged until the faulty capacitor is replaced. As long as there’s a short circuit on the board, the MCU won’t function either.

R8bert, 2024-10-20:
> If the capacitor is overloaded, it typically becomes conductive and causes a short circuit on the board. However, you won’t know if the MCU is also damaged until the faulty capacitor is replaced. As long as there’s a short circuit on the board, the MCU won’t function either.

I replaced all the capacitors and noticed that my cable (which shows power consumption in watts) wasn't lighting up. My initial thought was that something was still shorted and that my charger had gone into protection mode. Afterward, I checked the VCC and GND pins on the MCU using diode mode and found a short. I removed the MCU, and after doing so, my power LED turned on. I replaced the MCU, and everything seemed fine, but I haven’t programmed it yet. I’m currently waiting for 35V capacitors to arrive—maybe my 50V caps are just cheap and weak.
 Should I check anything else?

wagiminator, 2024-10-20:
I can really only imagine that there’s a problem with the LDO. The MCU could only have been damaged by overvoltage.

R8bert, 2024-10-20:
I used a HT7533-1 like in the BOM (link https://www.aliexpress.com/item/33015904481.html)
[image]
It is marked with 7533A-1. I will do some tests and see where the problem is, but i really dont know where to start :)))

R8bert, 2024-10-21:
@wagiminator I ran some tests on the HT7533-1 and observed that the voltage on the VOUT pin drops from the VIN reference of 24V down to 3.3V as expected. However, during the regulation process, the 24V is temporarily passed to the MCU, causing it to fry before the LDO fully stabilizes the voltage to 3.3V.

I'm not sure what to try next. Have you encountered any issues with the LDO using 20V PD?

R8bert, 2024-10-21:
https://youtu.be/0F6JsXu01Ms this is the time it takes to decrease the voltage to 3.6v from 21v :)))
The MCU is fried in that time:))

wagiminator, 2024-10-21:
I’ve never had any issues with these regulators. Keep in mind that the LDO requires suitable input and output capacitors to regulate the voltage properly. Still, the behavior you’re describing is really strange...

R8bert, 2024-11-01:
> I’ve never had any issues with these regulators. Keep in mind that the LDO requires suitable input and output capacitors to regulate the voltage properly. Still, the behavior you’re describing is really strange...

Hi. I just recieved the 35v caps (https://www.aliexpress.com/item/1005006049225016.html) and i just assambled a new pcb with those caps, from whai i saw, the regulating time was faster but still there it was some seconds (arround 5 seconds with +5v) and my mcu still got burned. I dont know what else to try, can you please provide a link with the caps you are using?

wagiminator, 2024-11-17:
Did you try a different LDO?