wagiminator/Development-Boards #10: All CH32V projects should have no framework set

maxgerhardt, 2024-11-15:
E.g., https://github.com/wagiminator/Development-Boards/tree/main/CH32X035F7P6_DevBoard/software/blink will still compile without
```ini
framework = noneos-sdk
board_build.use_lto = yes
board_build.use_builtin_startup_file = no
board_build.use_builtin_system_code = yes
board_build.use_builtin_debug_code = no
board_build.cpp_support = no
```

This removes the WCH SDK code from the build and prevents to cause problems where `use_builtin_debug_code = no` is problematic with the CH32X035 SDK (because one SDK file actually relies on that, see https://github.com/Community-PIO-CH32V/platform-ch32v/issues/75)

wagiminator, 2024-11-17:
Hi, and thanks for the hint! I'm a bit busy at the moment, but as soon as I have some free time, I'll try it out and make the changes.

wagiminator, 2024-11-24:
Works great, thank you!